wollewald/MCP23017_WE #1: 4 Rotary encoders

thijstriemstra, 2020-07-22:
@wollewald thanks for this library, it looks better (maintained) than the other alternatives.

I have 4 rotary encoders with internal push buttons wired to a mcp23017. This is how they're wired to the MCP23017:

| Rotary Encoder | Pin Label | MCP23017 | Pin Type |
| --- | --- | --- | --- |
| 1 | CLK | `7` | Digital Input |
| 1 | DT | `4` | Digital Input |
| 1 | Button | `3` | Digital Input |
| 2 | CLK | `5` | Digital Input |
| 2 | DT | `6` | Digital Input |
| 2 | Button | `2` | Digital Input |
| 3 | CLK | `9` | Digital Input |
| 3 | DT | `10` | Digital Input |
| 3 | Button | `0` | Digital Input |
| 4 | CLK | `8` | Digital Input |
| 4 | DT | `11` | Digital Input |
| 4 | Button | `1` | Digital Input |

I have an interrupt pin (19) and reset pin (23) connected to the ESP32. The interrupt pin connects to ITA on the MCP23017. With the following code I get serial prints for everything on port A but port B is not printed (e.g. caught by interrupt). Can you help figure out what's wrong? How do I read the state of B pins?

```cpp
#define MCP_ADDRESS 0x20 // (A2/A1/A0 = LOW) 

#include <Arduino.h>
#include <Wire.h>
#include <MCP23017.h>

int interruptPin = 19;
int resetPin = 23;

volatile bool event = false;
byte intCapReg; 

MCP23017 myMCP(MCP_ADDRESS, resetPin);

void IRAM_ATTR eventHappened(){
  Serial.println("event happened");
  event = true;
}

void setup(){ 
  pinMode(interruptPin, INPUT);
  attachInterrupt(digitalPinToInterrupt(interruptPin), eventHappened, RISING);

  Serial.begin(115200);
  Wire.begin(SDAPin, SCLPin);

  myMCP.Init();
  myMCP.setPortMode(B00000000, A);  // Port A: all pins are INPUT
  myMCP.setPortMode(B00000000, B);  // Port B: all pins are INPUT
  delay(1000);
  myMCP.setInterruptPinPol(HIGH);
  delay(10);
  // interrupt pins, port, DEFVALB
  myMCP.setInterruptOnDefValDevPort(B11111111, A, B00001111);
  myMCP.setInterruptOnDefValDevPort(B11111111, B, B00001111);
  // pull-up for A0-A3
  myMCP.setPortPullUp(B00001111, A);
  event = false;
  Serial.println("ready");
}

void loop(){ 
  intCapReg = myMCP.getIntCap(A);
  if (event) {
    byte intFlagReg, eventPin; 
    intFlagReg = myMCP.getIntFlag(A);
    eventPin = log(intFlagReg) / log(2);
    intCapReg = myMCP.getIntCap(A);
    Serial.println("Interrupt!");
    /*Serial.print("Interrupt Flag Register: ");
    Serial.println(intFlagReg, BIN); 
    Serial.print("Interrupt Capture Register: ");
    Serial.println(intCapReg, BIN);*/
    Serial.print("Pin No.");
    Serial.print(eventPin);
    Serial.print(" went ");
    if((intFlagReg&intCapReg) == 0){ // HIGH/LOW or LOW/HIGH change?
      Serial.println("LOW");
    }
    else{
      Serial.println("HIGH");
    }
    event = false;
    Serial.println();
  }
}
```

thijstriemstra, 2020-07-22:
> The interrupt pin connects to ITA on the MCP23017.

Should ITB be connected to the esp32 interrupt pin as well? [This library's example](https://github.com/prampec/MCP23017int/blob/master/examples/InterruptBasics/InterruptBasics.ino) states that `You can connect either INTA or INTB or both to the Arduino interrupt pin.` I can wire them together but this doesn't take care of the reading part in the code..

wollewald, 2020-07-22:
If you want to use only one interrupt pin you have to use the setIntMirror(val) function. The default val is 0 or OFF, which means ITA and ITB are seperate. With myMCP.setIntMirror(1) or myMCP.setIntMirror(ON) you can merge interrupt pins. But only the interrupt signal is merged. You will still have to read the capture and flag registers for both ports. Hope this helps - good luck.

thijstriemstra, 2020-07-22:
Thanks for the details @wollewald. When should I make the call to `setIntMirror`? Because adding it, after `myMCP.init()` for example, results in.. nothing, there is no input or interrupts called anymore. Since I'm saving potentially 16 pins on the ESP32 I don't mind using another pin for an additional interrupt. I'll give that a try and report back.

thijstriemstra, 2020-07-22:
I added another interrupt pin for ITB but the readings are all screwed up now. I'm probably not correctly parsing the results. The code looks like this:

```cpp
#include <Arduino.h>
#include <Wire.h>
#include <MCP23017.h>

#define MCP_ADDRESS 0x20 // (A2/A1/A0 = LOW)

int interruptPin1 = 19;
int interruptPin2 = 18;
int resetPin = 23;

volatile bool eventA = false;
volatile bool eventB = false;
byte intCapReg; 

MCP23017 myMCP(MCP_ADDRESS, resetPin);

void IRAM_ATTR eventAHappened() {
  eventA = true;
}

void IRAM_ATTR eventBHappened() {
  eventB = true;
}

void setup() {
  Serial.begin(115200);

  pinMode(interruptPin1, INPUT);
  pinMode(interruptPin2, INPUT);
  attachInterrupt(digitalPinToInterrupt(interruptPin1), eventAHappened, RISING);
  attachInterrupt(digitalPinToInterrupt(interruptPin2), eventBHappened, RISING);
  
  Wire.begin(SDAPin, SCLPin);

  myMCP.Init();
  myMCP.setPortMode(B00000000, A);  // Port A: all pins are INPUT
  myMCP.setPortMode(B00000000, B);  // Port B: all pins are INPUT

  delay(1000);
  myMCP.setInterruptPinPol(HIGH);
  delay(10);

  // interrupt pins, port, DEFVAL
  myMCP.setInterruptOnDefValDevPort(B11111111, A, B00001111);
  myMCP.setInterruptOnDefValDevPort(B11111111, B, B00001111);

  // pull-up for A0-A3
  myMCP.setPortPullUp(B00001111, A);

  eventA = false;
  eventB = false;
  Serial.println("ready.");
}

void getValue(enum MCP_PORT port) {
  byte intFlagReg, eventPin; 
  intFlagReg = myMCP.getIntFlag(port);
  eventPin = log(intFlagReg) / log(2);
  intCapReg = myMCP.getIntCap(port);

  Serial.print(millis());
  Serial.print(" - ");
  Serial.print("Pin ");
  Serial.print(eventPin);
  Serial.print(" went ");

  if ((intFlagReg&intCapReg) == 0) { // HIGH/LOW or LOW/HIGH change?
    Serial.println("LOW");
  }
  else{
    Serial.println("HIGH");
  }
}
void loop() { 
  intCapReg = myMCP.getIntCap(A);
  if (eventA) {
    getValue(A);
    eventA = false;
  }
  intCapReg = myMCP.getIntCap(B);
  if (eventB) {
    getValue(B);
    eventB = false;
  }
}
```

How do I properly read out A and B with 2 interrupt pins since `setIntMirror` doesn't seem to do anything?

wollewald, 2020-07-23:
At first sight the code looks Ok. Basically that's how I also would have done this. If you use only one port it is working? I will have to look more in detail, but it's my first working day after holidays and therefore I might not be able to do it today.

thijstriemstra, 2020-07-23:
> If you use only one port it is working?

All pins on port A are working, nothing on B is working. I'm wasting so much time on this, dammnit. I thought it might be related to the global Wire instance that is being used, maybe the ability to pass in a specific TwoWire instance (since I use different SDA/SCL pins) might help.

thijstriemstra, 2020-07-23:
I tried the read gpio example and finally some progress, the status on GPIO B ports is visible in the prints when rotating one of the encoders on the GPIOB section.

```
Status GPIO A: 1111
Status Port A, Pin 3: 1
Status GPIO B: 1001
Status Port B, Pin 0: 1
-------------------------------------
Status GPIO A: 1111
Status Port A, Pin 3: 1
Status GPIO B: 1001
Status Port B, Pin 0: 1
-------------------------------------
Status GPIO A: 1111
Status Port A, Pin 3: 1
Status GPIO B: 1000
Status Port B, Pin 0: 0
```

wollewald, 2020-07-23:
Sorry that you have to spend so much time. I have just set up the MCP23017 on a breadboard and added pulldown resistors to all pins on port A and B. Then I took the interrupt on change example sketch (which was designed to monitor port B) and changed it quick and dirty to also monitor port A. 

With this setup I can detect interrupts on both ports, when I connected pins to a HIGH Signal. I also tried the Mirror function by commenting one of the attachInterrupt functions and uncommenting the line after myMCP.Init(). This was also working - if you want to use this, some code needs to be added to check if the interrupt was on port A or B. 

But the issue is that I can't see any bug in your sketch. Strange. 

```
#define MCP_ADDRESS 0x20 // (A2/A1/A0 = LOW) 
#include <Wire.h>
#include <MCP23017.h> 
int interruptPin1 = 2;
int interruptPin2 = 3;
volatile bool eventA; 
volatile bool eventB; 
byte intCapReg; 

MCP23017 myMCP(MCP_ADDRESS,5); // 5 = ResetPin

void setup(){ 
  pinMode(interruptPin1, INPUT);
  pinMode(interruptPin2, INPUT);
  attachInterrupt(digitalPinToInterrupt(interruptPin1), eventAHappened, RISING);
  attachInterrupt(digitalPinToInterrupt(interruptPin2), eventBHappened, RISING);
  Serial.begin(9600);
  Wire.begin();
  myMCP.Init(); 
  //myMCP.setIntMirror(ON);
  delay(1000);
  myMCP.setInterruptPinPol(HIGH); // set INTA and INTB active-high
  delay(10);
  myMCP.setInterruptOnChangePort(B11111111, A);
  myMCP.setInterruptOnChangePort(B11111111, B); //set all B pins as interrrupt Pins
  eventA=false;
  eventB=false;
}  

void loop(){ 
  intCapReg = myMCP.getIntCap(A);
  intCapReg = myMCP.getIntCap(B); // ensures that existing interrupts are cleared
  if(eventA){
    //delay(200);
    byte intFlagReg, eventPin; 
    intFlagReg = myMCP.getIntFlag(A);
    eventPin = log(intFlagReg)/log(2);
    intCapReg = myMCP.getIntCap(A);
    Serial.println("Interrupt!");
    Serial.print("Interrupt Flag Register: ");
    Serial.println(intFlagReg,BIN); 
    Serial.print("Interrupt Capture Register: ");
    Serial.println(intCapReg,BIN); 
    Serial.print("Pin No.");
    Serial.print(eventPin);
    Serial.print(" went ");
    if((intFlagReg&intCapReg) == 0){  //LOW-HIGH or HIGH-LOW interrupt?
      Serial.println("LOW");
    }
    else{
      Serial.println("HIGH");
    }
    eventA = false; 
  }
  if(eventB){
    //delay(200);
    byte intFlagReg, eventPin; 
    intFlagReg = myMCP.getIntFlag(B);
    eventPin = log(intFlagReg)/log(2);
    intCapReg = myMCP.getIntCap(B);
    Serial.println("Interrupt!");
    Serial.print("Interrupt Flag Register: ");
    Serial.println(intFlagReg,BIN); 
    Serial.print("Interrupt Capture Register: ");
    Serial.println(intCapReg,BIN); 
    Serial.print("Pin No.");
    Serial.print(eventPin);
    Serial.print(" went ");
    if((intFlagReg&intCapReg) == 0){  //LOW-HIGH or HIGH-LOW interrupt?
      Serial.println("LOW");
    }
    else{
      Serial.println("HIGH");
    }
    eventB = false; 
  }
}

void eventAHappened(){
  eventA = true;
}
void eventBHappened(){
  eventB = true;
}
```

thijstriemstra, 2020-07-24:
thanks for the feedback @wollewald. For time sake I decided to ditch the MCP23017 for this project and connect the encoders directly to the ESP32. Found a nice library that seems to work fine so far: https://github.com/madhephaestus/ESP32Encoder

I will revisit the MCP23017 later on and give your example a try and let you know. thanks!

thijstriemstra, 2020-07-24:
Also, I would suggest to add a method that returns the pin and value of last interrupt, basically this:

```
byte intFlagReg, eventPin; 
    intFlagReg = myMCP.getIntFlag(A);
    eventPin = log(intFlagReg)/log(2);
    intCapReg = myMCP.getIntCap(A);
    Serial.println("Interrupt!");
    Serial.print("Interrupt Flag Register: ");
    Serial.println(intFlagReg,BIN); 
    Serial.print("Interrupt Capture Register: ");
    Serial.println(intCapReg,BIN); 
    Serial.print("Pin No.");
    Serial.print(eventPin);
    Serial.print(" went ");
    if((intFlagReg&intCapReg) == 0){  //LOW-HIGH or HIGH-LOW interrupt?
      Serial.println("LOW");
    }
    else{
      Serial.println("HIGH");
    }
    eventA = false; 
  }
```

It's a lot of 'odd' code for  a user that could be abstracted away into a good library method, no?

wollewald, 2020-07-24:
Thanks for this recommendation!

closed 2020-07-24 by wollewald