wollewald/MCP23017_WE #3: Compile warnings

thijstriemstra, 2020-07-27:
See travis build: https://travis-ci.com/github/wollewald/MCP23017_WE/jobs/365247364#L419

```
lib/MCP23017_WE/src/MCP23017.cpp: In member function 'byte MCP23017::getIntFlag(MCP_PORT)':

lib/MCP23017_WE/src/MCP23017.cpp:386:1: warning: control reaches end of non-void function [-Wreturn-type]

 }

 ^

lib/MCP23017_WE/src/MCP23017.cpp: In member function 'byte MCP23017::getPort(MCP_PORT)':

lib/MCP23017_WE/src/MCP23017.cpp:404:1: warning: control reaches end of non-void function [-Wreturn-type]

 }

 ^

lib/MCP23017_WE/src/MCP23017.cpp: In member function 'byte MCP23017::getIntCap(MCP_PORT)':

lib/MCP23017_WE/src/MCP23017.cpp:409:1: warning: control reaches end of non-void function [-Wreturn-type]

 }

 ^
```

wollewald, 2020-07-27:
Will have a look at this, but this will now take some days. And I prefer to amend things in one go. I am sure you'll find more.

wollewald, 2020-07-30:
This is solved now.

closed 2020-07-30 by wollewald