wollewald/MCP23017_WE #4: Reset optional?

thijstriemstra, 2020-08-18:
I just found out the MCP203017 reset pin was hooked up to a different pin on the ESP32 than one I specified. But it doesn't seem to have made a difference, everything properly worked anyway. So is the reset pin optional?

wollewald, 2020-08-18:
The reset function is only used by the Init function to ensure that the MCP23017 starts with a defined status. Otherwise, if you reset the ESP32, you might not reset the MCP23017. And then it could still "remember" its status before the ESP32 was reset and cause problems. And of course you need the reset pin if you want to use the reset function.

thijstriemstra, 2020-08-18:
But powering down the ESP32 and MCP23017 will reset the MCP23017 right? Unlike the ESP32 reset button?

wollewald, 2020-08-18:
Yes that's right.

thijstriemstra, 2020-08-18:
oh closed, too soon. For now I modified the library to not change the `pinMode` for the reset pin (e.g. when i pass in -1) but it would be nice to make the reset pin officially optional.

wollewald, 2020-08-19:
Please try. You can use now:
`MCP23017 myMCP(MCP_ADDRESS)` 
or
`MCP23017 myMCP(MCP_ADDRESS, pinNumber)`
All instructions in Init should ensure that the status of the MCP23017 is defined, even if not reset.

thijstriemstra, 2020-08-20:
awesome, thanks!

closed 2020-08-20 by thijstriemstra