wollewald/MCP23017_WE #5: How to change I2C address

thijstriemstra, 2020-10-07:
hi @wollewald I have one of these mcp23017 devices:

[image]

and the address 0x20 shows up. I want to add another one and change the address to let's say 0x21. Any idea what I need to do with these A0, A1, A2 jumpers?

wollewald, 2020-10-07:
Hi @thijstriemstra, the address scheme is in binary number: 1 0 0 A2 A1 A0. If Ax (with x = 0, 1, 2) is connected to GND replace Ax it by zero, if connected to VCC replace it by one. On the photo all Ax are connected to GND, so the address is 100000 = 0x20. Connect A0 to VCC and leave the rest as is and you'll get the address 100001 = 0x21. --> and so on...
So for 0x21 solder out the SMD resistor and connect the middle pad with VCC. The resistor is 0 Ohm, so you can also connect it with a drop of  tin-solder.

thijstriemstra, 2020-10-07:
Awesome, thanks for the confirmation, I'll give it a try.

thijstriemstra, 2020-10-07:
ps. i couldn't find it in your article, maybe useful to add that info there as well.

thijstriemstra, 2020-10-07:
```
Scanning...
I2C device found at address 0x21  !
done
```

thx!

closed 2020-10-07 by thijstriemstra