wollewald/MCP23017_WE #7: Invert pin state

bobcroft, 2020-12-07:
Many of the opto isolated relay banks are active low requiring a switch to ground to turn them on.  I have begun using your flexible and versatile library, however it is counter-intuitive or seems illogical to set the output state in code to 'OFF' so that the output pin goes low to turn ON the relay.  Is it possible to have an 'invert' function so issuing an  'ON' command sets the output to low on the MCP23017.

wollewald, 2020-12-08:
@bobcroft, the logic of the library follows the logic of the Arduino or microcontrollers in general, where HIGH = 1 = ON and LOW = OFF = 0. ON and HIGH are defined to be 1 and OFF and LOW are defined to be 0 by a preprocessor  directive (#define) in the Arduino libraries, so that is not my idea. 

That means every ON, HIGH, LOW, OFF is replaced by 1 or 0 by the preprocessor. The consequence is I can't change ON and OFF in isolation. Or in other words: if I change ON and OFF for some functions such as setPin I would change HIGH an LOW and 0 and 1. setPin(x, A, LOW) as an example would set pin x to HIGH state. And to me this is also not logic.

Here's my proposal. You add something like:
#define RELAY_ON 0
#define RELAY_OFF 1
to your code. Then set setPin(x, A, RELAY_0N) would switch your relay on as needed.

bobcroft, 2020-12-08:
Wolfgang,

 

Thank you for your comprehensive reply which is informative and clears up some things in my mind.  I had wondered if your suggested method (proposal) would work.  I’ll try it today.

 

Thank you for an excellent library.  I am using it on a ESP8266 with platformIO.

 

Bob

wollewald, 2020-12-24:
@bobcroft , I will now close this issue. I hope my proposal worked for you.

closed 2020-12-24 by wollewald

bobcroft, 2020-12-24:
Wolfgang,

 

Yes, it did work, thank you very much.  Your library is now my go to library for the ESP boards.

 

Merry Christmas and a Happy New Year.

 

Bob

wollewald, 2020-12-24:
Great thanks for the feedback.And also Merry Christmas and Happy new Year for you!Let's hope for a less strange 2021.--Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.