wollewald/MCP23017_WE #8: GPA7 and GPB7 not work as input

kang2k10, 2021-04-28:
My board, all pins GPA and GPB4-GPB7 as input, but in testing GPA7 and GPB7 not change state.

wollewald, 2021-04-28:
@kang2k10 difficult for me to comment on this without more details. I think if this would be common issue or even a bug I would have heard a bout this before. Can you provide more details, e.g. the code you applied when you experienced the issue?

kang2k10, 2021-04-29:
Sorry. i use example from library, with minor change:

#define MCP_ADDRESS1 0x20
#define MCP_ADDRESS2 0x21 
#include <Wire.h>
#include <MCP23017.h>
byte portStatus;

/*
   You can choose if you want to use the reset function or not.
   If you don't need it, you can save one pin.
*/
MCP23017 myMCP1(MCP_ADDRESS1); 
MCP23017 myMCP2(MCP_ADDRESS2); 

void setup() {
  Serial.begin(115200);
  Wire.begin();
  myMCP1.Init();
  myMCP2.Init();
  myMCP1.setPortMode(B00000000, A); 
  myMCP1.setPortMode(B00001111, B); 
  myMCP2.setPortMode(B00000000, A);  
  myMCP1.setPortPullUp(B11111111, A);
  myMCP2.setPortPullUp(B11111111, A);
  myMCP1.setPortPullUp(B11110000, B);
}

void loop() {
  portStatus = myMCP1.getPort(A);
  Serial.println(portStatus, BIN);

}

All pin except GPA7 change status, whith button press.

wollewald, 2021-04-29:
Hi @kang2k10 , I have just tried the following sketch with only one MCP23017:

````
#define MCP_ADDRESS1 0x20

#include <Wire.h>
#include <MCP23017.h>
byte portStatus;

MCP23017 myMCP1(MCP_ADDRESS1);

void setup() {
Serial.begin(115200);
Wire.begin();
myMCP1.Init();
myMCP1.setPortMode(B00000000, A);
myMCP1.setPortMode(B00001111, B);
myMCP1.setPortPullUp(B11111111, A);
myMCP1.setPortPullUp(B11110000, B);
}

void loop() {
  portStatus = myMCP1.getPort(A);
  Serial.println(portStatus, BIN);
}
````

A least this one works as expected. 

Be aware that the preceding zero will not be displayed. So if you connect GPA6 to GND, the portStatus will be 10111111. If you connect GPA7 to GND the portStatus will be 1111111. 

If it's not that simple issue I would ask you to try my sketch and provide feedback if it works on your side.

kang2k10, 2021-04-29:
Yes, I upload you sketch. And if i connect GPA6 to GND, port status 10111111.
If i connect GPA7 to GND, the port status is 11111111.
How determine change status on pin GPA7 ?

wollewald, 2021-04-29:
So if you connect GPA7 it's really 11111111 (so 8 times 1) and not 1111111 (7 times 1)? Sorry for asking again, I want to be sure  Then it becomes difficult, because if I don't see the same issue like you, I won't be able to solve it.

kang2k10, 2021-04-29:
**#facepalm**
i, m sorry. realy, i change sketch
#define MCP_ADDRESS1 0x20

#include <Wire.h>
#include <MCP23017.h>
byte portStatus, portStatus_old;

MCP23017 myMCP1(MCP_ADDRESS1);

void setup() {
  Serial.begin(115200);
  Wire.begin();
  myMCP1.Init();
  myMCP1.setPortMode(B00000000, A);
  myMCP1.setPortMode(B00001111, B);
  myMCP1.setPortPullUp(B11111111, A);
  myMCP1.setPortPullUp(B11110000, B);
}

void loop() {
  portStatus = myMCP1.getPort(A);
  if (portStatus != portStatus_old)
  {
    Serial.println(portStatus, BIN);
    portStatus_old = portStatus;
  }
}
and see, if i connect GPA7 to GND, status 1111111, not 11111111

Sorry, and verry thanks!

wollewald, 2021-04-29:
No worries, I am happy that it works!! :-)

closed 2021-04-29 by wollewald