wollewald/MCP23017_WE #9: Works for Arduino?

paulrose71, 2021-04-30:
Hi there, do your examples work with Arduino? I've just tried using one of the examples but it couldnt compile in my Arduino IDE? Thanks for any help! Paul

wollewald, 2021-04-30:
Hi Paul @paulrose71 , yes I have devloped the library using the Arduino UNO. So it should work.

Do you have installed other libraries for the MCP23017? And if yes, do these contain a file called MCP23017.h? If yes, can you delete the libraries or - in case you just want to test my library - move the other library somewhere else? Atleast they should not stay in the library folder. 

That was the first library I have written. For newer libraries I have learned and added a "_WE" to the file names to avoid such issues. 

If this is not the solution to the issue I would ask you to provide the compiler messages. You get most information if go into File -> Preferences, choose "All" for Compiler Warnings and tick the boxes behind "Show verbose output during:".  Best wishes, Wolfgang

paulrose71, 2021-04-30:
Thanks alot Wolfgang! I deleted all the other libraries and just kept your _WE library. Its compiling fine now. Thanks again. I would have never worked that out! Paul

wollewald, 2021-04-30:
Great that it works. When I first had this issue it took me hours to find out.

closed 2021-04-30 by wollewald