wollewald/MCP23017_WE #10: How to use an AC solid state relay to also switch DC?

paulrose71, 2021-05-07:
Hi Wolfgang,

I'm using a 16 channel AC solid state relay board to switch on and off 24VAC, 0.5A loads. That's all fine. Now I'd like to be able to use the relays to switch on and off a 5VDC, 50mA load too! Of course, I can use these relays to turn on the 5VDC. But I understand 
I can't use them to turn off the 5VDC. Sooo, I was wondering, if I reset the relay board- via the pin 5 reset, would that turn off all the relays? And therefore turn off the 5VDC, 50mA load?

Thanks alot,
Paul

paulrose71, 2021-05-07:
Could I connect pin 5 on port A to my Arduino (say DIO pin 5 on my Arduino)? Then to do a reset, I tell my Arduino pin 5 to go low, wait, then high again? 

Thanks, Paul

paulrose71, 2021-05-10:
Hi Wolfgang,

This is what I have assembled. I have a Lora tx unit and a Lora rx unit......

Hardware:
My laptop with Arduino IDE
Arduino UNO R3 with a Lora shield on top of it (Dragino, 433MHz), transmitter unit, with its keypad and 1.3" OLED.
Arduino UNO R3 with a Lora shield on top of it (Dragino, 433MHz), receiver unit
KRIDA Electronics 16 channel solid state i2c relay board (MCP23017), attached to receiver unit

Power:
Arduino receiver: USB cable from laptop
Arduino transmitter: 12VDC cord from a 240VAC to 12VDC transformer, which is powered from main house supply (240VAC).
Relay board: 5VDC supplied from a small converter (24VAC to 5VDC). The converter is supplied from a transformer (240VAC to 24VAC). The transformer is supplied from main house supply (240 VAC).

Receiver end wiring:
Lora shield SCL to relay board SCL
Lora shield SDA to relay board SDA
Lora shield GND to relay board GND
Relay board VCC to small converter +DC
Relay board GND to small converter -DC

I did a small test using my multimeter on the relay board to see what happens when I turn on and turn off the 5VDC. When I turn on the relay the 5VDC arrives. But when I turn it off, the 5VDC is still there! The relay doesn't actually turn off. But when I press the reset button on the Arduino, the relay board also resets, and thats when the 5VDC stops. So, the only way to turn off the 5VDC is by resetting the relay board. And not by "manually" turning  off the relay. Below is an extract from the OMRON website.....

https://www.omron.com.au/service_support/FAQ/FAQ02101/index.asp

Question
Can a DC load be used with Solid-state Relays for AC loads?

Answer
No, it cannot be used. The element characteristics of Solid-state Relays will not allow it to reset if a DC load is used.

Explanation
Solid-state Relays for AC load switching use thyristors and triacs as output elements. By turning Solid-state Relays input ON, the output can be turned ON, but simply turning the input OFF doesn't mean that the output will be turned OFF. This is because Solid-state Relays have a characteristic that it stays turned ON until the load current becomes zero regardless of the absence of input signals into Solid-state Relays.
The DC current will never become zero, therefore it will not be reset.

Regards,
Paul

wollewald, 2021-05-10:
Hi Paul, somehow I have overlooked the "A" in VAC....

To reset the MCP23017 its reset pin needs a "LOW" signal. While you apply the "LOW" it stays in reset mode. Only when you change to HIGH it will work again. My library has a reset function which can do this for you. If your objsect is named myMCP, you just call myMCP.reset() . That's all. In the backgrund it applies a LOW for 10 milliseconds, the changes to high again. 

If a reset of the MCP23017 will successfully reset the relay is something you have to find out. A reset will turn all register in default state. For the GPIOs it means, that they will be all INPUT/LOW.  Regards, Wolfgang

paulrose71, 2021-05-12:
OK, perfect. I'll do that then. Thanks alot Wolfgang. Its appreciated. Paul

wollewald, 2021-05-15:
I close this issue now.

closed 2021-05-15 by wollewald