wollewald/MCP23017_WE #12: Support for MCP23S17

thijstriemstra, 2021-08-05:
From https://www.best-microcontroller-projects.com/mcp23017.html:

> the MCP23S17 uses the SPI interface that can operate at 10MHz (a lot faster than the I2C version). This SPI device has the same number and arrangement of pins, but uses two unused (I2C) pins to implement the SPI interface. In all other respects it operates the same as the MCP23017.

[image]

thijstriemstra, 2021-08-05:
If you want I can send you such a chip @wollewald

wollewald, 2021-08-06:
@thijstriemstra , that is a login extention to the library. Thanks for your offer to send a MCP23S17. I ordered some for a few Euro in China. I am not sure when I will find some time to do that. Might take a while.

wollewald, 2021-09-09:
Hi @thijstriemstra , I have now added the support for the MCP23S17. You find an example sketch and the wiring. I tested it on an Arduino Nano, but should also work on an ESP32 - please try.

thijstriemstra, 2021-09-10:
awesome @wollewald! im on holiday and will be able to test in a week or so. Did you notice/test any difference in speed since the SPI version should be much faster?

wollewald, 2021-09-10:
@thijstriemstra , it should be much faster but I did not test speed. I just controlled some LEDs. Enjoy your holiday.

wollewald, 2021-09-10:
I just did a speed test on an Arduino Nano. I toggled one Pin at highest speed, that means without any delay in the loop. I have measured the frequency (from HIGH to HIGH or LOW to LOW)

MCP23017 (I2C): 0.746 kHz (standard 100 kHz I2C frequency)
MCP23017 (I2C): 1.92 kHz  (400 kHz I2C frequency, using Wire.setClock(400000) )
MCP23S17 (SPI): 11.9 kHz  (4 MHz clock speed (default))
MCP23S17 (SPI): 14.0 kHz  (8 MHz clock speed, using SPI.setClockDivider(SPI_CLOCK_DIV2);)

For I2C 400kHz is the maximum with an Arduino. According to the data sheet the MCP23017 can work with 1.7MHz, but the MCU needs to support it. It may be possible with an ESP32. 

The maximum SPI clock speed for the MCP23S17 is 10 MHz. Interestingly the speed increase was not as big as expected changing from 4 MHz SPI clockspeed to 8 MHz. Maybe in this region the bottleneck is elsewhere. Maybe I try with an ESP32.

wollewald, 2022-03-18:
@thijstriemstra Hi, I added now also a function which allows to set the SPI clock speed and close this issue.

closed 2022-03-18 by wollewald

thijstriemstra, 2022-06-27:
hi @wollewald, great article on https://wolles-elektronikkiste.de/en/mcp23x1y-port-expander!

I'm trying to adapt the example to an esp32-c3-devkitm-1 board and it wasn't totally clear how to hook it up.

This part helped though:

> Accordingly, it has an SCK (Serial Clock), SI (Slave In), SO (Slave Out) and a CS (Chip Select) pin.

The board uses:

```c++
#define SPI_MOSI 6
#define SPI_MISO 5 
#define SPI_SCLK 4 
#define SPI_CS 7
#define SPI_N 1
```

So I'll connect:

- mcp23s17 SCK -> esp32 SPI_SCLK
- mcp23s17 SI - > esp32 SPI_MOSI
- mcp23s17 SO -> esp32 SPI_MISO
- mcp23s17 CS -> esp32 SPI_CS
- mcp23s17 RST -> any esp32 digital pin

And I'll use 3.3V instead of 5V used in the arduino example since the esp32 uses 3.3v pins (does this make sense, is this actually needed?)

A reference to MOSI/MISO/SCLK in the article and/or examples would be useful to identify the correct pins on alternative boards like this esp32.

wollewald, 2022-06-27:
Hi @thijstriemstra, the connection looks OK to me. As I wrote in one of the other issues you can connect the reset to HIGH level and pass a (virtual) pin >= 99. 
And a voltage of 3.3V is no issue. The mcp23s17 can deal with 1.8 to 5.5 volts. 
If it doesn't work with the connections you have chosen, then I have to try myself. But it seems this board is currently hardly available at resonable prices.

thijstriemstra, 2022-06-27:
It seems the board isn't properly supported by platformio yet, and I tried with the board below and everything works great!

[image]

thijstriemstra, 2022-06-27:
oh, are interrupts supported yet on mcp23s17?