wollewald/MCP23017_WE #15: setPinMode vs setPortMode -> INPUT_PULLUP

pascaltippelt, 2021-12-21:
Hi!

First thanks for your incredible work. I tried the Adafruit library, but for my setup (ESP-01s + 2x MCP23017) it did not work. Your library **does work great**, but I have an understanding-questing:

Using _setPortMode_ I can set a port to input or output, correct? But can I set it to INPUT_PULLUP too, or is the only way to do this the _setPinMode_ function?

Best

Pascal

wollewald, 2021-12-21:
Hi @pascaltippelt,

yes, with setPortMode you can set the INPUT/OUTPUT mode of all pins of a Port. 

With:
- setPortMode(B11111111, A) you would set all pins of port A to OUTPUT
- setPortMode(B00000000, A) you would set all pins of port A to INPUT
- setPortMode(B11110000, A) you would set pins 0-3 of port A to INPUT, and 4-7 to OUTPUT

With
- setPortPullUp(B11110000, A) you would set the PULLUP of Pins 4-7

So if you use:
- setPortMode(B11110000, A)
and 
- setPortPullUp(B00001111, A) 
pins 0-3 would be INPUT_PULLUP. 

With setPinMode you can set INPUT_PULLUP using only that one command, but only for a single pin. 

Hope this answers the question and does not confuse you completely!

pascaltippelt, 2021-12-22:
Hi @wollewald ,
that clearifys it. Thank you very much!

closed 2021-12-22 by pascaltippelt