wollewald/MCP23017_WE #18: v1.6.1 breaks something

thijstriemstra, 2022-10-31:
my mcp23s17 + mcp23017 esp32 application seems to crash on v1.6.1, 1.6.0 seems to work fine.

Crash dump (if it helps at all):

```
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d28d8  PS      : 0x00060a30  A0      : 0x800d29f9  A1      : 0x3ffb2720  
A2      : 0x1119010a  A3      : 0x000f4240  A4      : 0xffffffff  A5      : 0x00060a23  
A6      : 0x00060a20  A7      : 0x00000001  A8      : 0x00000001  A9      : 0x3ff44000  
A10     : 0x00000000  A11     : 0xfffc1fff  A12     : 0x00060a20  A13     : 0x00060c23  
A14     : 0x007bdc20  A15     : 0x003fffff  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x11190126  LBEG    : 0x400d49b1  LEND    : 0x400d49c0  LCOUNT  : 0x00000000  

Backtrace:0x400d28d5:0x3ffb27200x400d29f6:0x3ffb2750 0x400d2a44:0x3ffb2770 0x400d2b35:0x3ffb2790 0x400d2cf2:0x3ffb27b0 0x400d1924:0x3ffb27d0 0x400d19b7:0x3ffb2800 0x400d553a:0x3ffb2820 

ELF file SHA256: 0000000000000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
```

wollewald, 2022-10-31:
@thijstriemstra , sorry for the inconvenience! I found and fixed the issue. The default SPI object ( _spi{&SPI} ) was missing in one of the constructors for SPI. Despite this bug, it strangely worked on an Arduino Nano and therefore I had not found it when I tested version 1.6.1. Interesting learning - it did not even produce a compiler warning on the Arduino.

1.6.1: MCP23017(uint8_t cs, uint8_t rp, uint8_t addr) : SPI_Address{addr}, resetPin{rp}, csPin{cs}, useSPI{true} {}
1.6.2: MCP23017(uint8_t cs, uint8_t rp, uint8_t addr) : **_spi{&SPI}**, SPI_Address{addr}, resetPin{rp}, csPin{cs}, useSPI{true} {}

This library is quite challenging to test when changes are applied. There so many different combinations of MCUs, MCP23x1y types, example sketches and options for passing SPI / TwoWire objects, using TinyWireM and using the reset pin or not. 

I have tested the new version 1.6.2 on an ESP32, Arduino Nano, including the option to use several MCP23S17 with different addresses.

Thanks for raising the issue.

thijstriemstra, 2022-10-31:
> I have tested the new version 1.6.2 

Once it's available in the platformio registry I'll give it a try (it's still on 1.6.1 for some reason).

> This library is quite challenging to test when changes are applied. There so many different combinations of MCUs, MCP23x1y types, example sketches and options for passing SPI / TwoWire objects, using TinyWireM and using the reset pin or not.

You can use the github actions with platformio setup for this: it allows you to test a configuration (e.g. an .ino file for mcp23s17 with spi) against one or more boards. Without such a setup you'll keep running into bugs like this IMO.

thijstriemstra, 2022-10-31:
> Once it's available in the platformio registry 

Tested 1.6.2 and works fine.

closed 2022-10-31 by thijstriemstra