wollewald/MCP23017_WE #21: random pins are driven

feaver12, 2023-02-02:
when i set a pin as a output and try to drive the pin on a event from lvgl gui button press i.e:
void turnoncentrelights() {
    myMCP.setPin(7, A, HIGH); // Send high signal to turn on relay
    Serial.println("centre lights turned on");
}
The chip drives just any random pin on port A, switching whichever relay it pleases, and not the one specified. i have 4.7k resistors on sda/scl and have wired RST pin to 3v3 with 1k resistor. using addr 0x27

wollewald, 2023-02-02:
Hi @feaver12, I don't know what a "lvgl gui button press" is, but I know for sure that the setPin() function works reliably. It must be something related with other parts of your code, hardware setup or something else.
So, not sure how I can help. Maybe I would be able to find the reason for the issue if you provide the complete code and more details about the circuit. 

When I have issues in a project the thing I do is that I reduce complexity. So in your case I recommend you take the MCP23017, your microcontroller and one LED and check if the myMCP.setPin(7,A,HIGH) command will switch on the LED. If this does not work you would have much simpler system with less parameters and it will be easier to find the issue.

wollewald, 2023-02-26:
Hi @feaver12 , are you still experiencing this issue? Or have you tried what I recommended?

wollewald, 2023-06-08:
@feaver12 Since there was no feedback since few months I close the issue.

closed 2023-06-08 by wollewald