wollewald/MCP23017_WE #23: Multiple MCP23017 outputs don't always work

pwolfman, 2023-07-24:
I have a project that has 3 MCP23017 connected and was working fine, until I upgrade to V1.6.5.  Sometimes when I compile my program all 24 output ports work fine and sometimes ports 9-24 will not work but output ports 1-8 always work.  All 24 inputs work 100% of the time.  So the second and third connected MCP output A ports don't always work.  I downgraded to V1.6.0 and that seems to work all the time.   I have tested the code on 2 different PCBA and both exhibit the same symptoms.  All outputs are on port A of the MCP.

wollewald, 2023-07-24:
It will take a few days until I will have access to my computer. Then I will have a closer look. Can you tell me which reset method you use, software (reset pin  >= 99) or hardware? Or can you publish the relevant code you use to create the mcp23017 objects?

wollewald, 2023-07-25:
Hi @pwolfman , the problems that do not have to be solved by me are my favorite ones 😀.

Interesting that it worked with former versions.

closed 2023-12-24 by wollewald