wollewald/MCP23017_WE #25: MCP23S17/MCP23017 missing two input bit of the 16Bit Chip

dev-markus, 2023-10-28:
Problem with follow Chips:
* MCP23S17
* MCP23017

when one of the pins (GPA7 or GPB7) are high there will output a strange result:

  correct output:
  00000000 00000000 (all pins are connected to a other input source at low)
  01000000 00000000 (one input are high, correct result)

  not correct result:  
  0000000 00000000  (one bit are missing from bank A or B)
  
why does the output show results where a 1/0 is missing?

Code:
`#include <SPI.h>
#include <MCP23S17.h>
#define CS_PIN 7   
#define RESET_PIN 5 
#define MCP_ADDRESS 0x20

MCP23S17 myMCP = MCP23S17(CS_PIN, RESET_PIN, MCP_ADDRESS);

int wT = 500;
byte portStatusA;
byte portStatusB;

void setup(){ 
 
  Serial.begin(9600);
  SPI.begin();

  if(!myMCP.Init()){
    Serial.println("Not connected!");
    while(1){} 
  }
 
  Serial.println("CHIP: ready");
  delay(wT);

  myMCP.setPortMode(0b00000000, A);   // Port A: all pins are INPUT
  myMCP.setPortMode(0b00000000, B);   // Port B: all pins are INPUT
  delay(wT);
  
  myMCP.setPortPullUp(0b11111111, A);  // pulled up
  myMCP.setPortPullUp(0b11111111, B);  // pulled up
}

void loop(){ 

  portStatusA = myMCP.getPort(A); // query the complete port status 
  portStatusB = myMCP.getPort(B); // query the complete port status 

  Serial.print(portStatusA, BIN);
  Serial.print(" ");
  Serial.print(portStatusB, BIN);
  Serial.println("");
 
  delay(100);
} 

/*
  Console results:

  Correct Outpu:
  00000000 00000000 (all pins are connected to a other input source at low)
  01000000 00000000 (one input are high, correct result)
  
  0000000 00000000  (one bit are missing when GPA7 or GPB7 at high)
*/
`

wollewald, 2023-10-29:
I just tried your sketch without any changes. My board was an Arduino Nano. I connected all GPIOs of the MCP23S17 to LOW (GND) except A6, which was connected to HIGH (VCC). That should be the same like you did. By the way, "01000000 00000000" would mean that A6 is HIGH and not A7.

My result is as expected:

[image]

And it also works with A7 HIGH and the rest LOW. So, I can't reproduce the issue. 

At least with an MCP23**S**17 it should also work on your side. For the MCP23**0**17 there is one known issue with GPIOA7 and B7. Microchip has changed the design and as a result these pins can only be used as outputs.

Not sure what else I can say or do. Maybe you can provide some more details of your microcontroller and the circuit?

dev-markus, 2023-10-29:
We had previously try the MCP23017, this have the problem with the two input pins. So we ordered the MCP23S17, this has apparently the same problem.

Will post more info (schematic, prototype) today, maybe its a problem with the circuit.

wollewald, 2023-10-29:
Preceding zeros are not displayed, i.e.:  0b01111111 = 0b1111111. So, if you push the button connected to at GPA7 you should get:

1111111 11111111  or if you print in decimal numbers: 127 255 or in hex: 0x7F 0xFF

Did you expect:

01111111 11111111 ? 

Or did you expect:

11111110 11111111 ? 

The latter is what you get if you pull GPA0 to LOW.

dev-markus, 2023-10-29:
would have imagined i get: "01111111 11111111".
in this case the mystery has been solved, chips are working correctly :-)
Thank you for your time!

wollewald, 2023-10-29:
Great! Happy to hear that. 

And it had a good side effect because while working on the issue I have noticed that I had a typo in one of the file names of the examples sketches. And therefore it was not displayed as example in the Arduino IDE.

closed 2023-10-29 by wollewald