wollewald/MCP23017_WE #27: How to check if MCP23017 is connected and responsive?

RemyDavies, 2023-12-24:
Hi wollewald, this is more just a question, not an issue.

I have a sketch which checks all I2C slave devices every 10 seconds, to ensure they are connected and responsive. 
If any I2C device does not respond, I then raise an alarm flag for that device.

Is there a library function for this purpose, to check if the MCP23017 is responsive?

I have tried the following:
```
  if ((mcp_0x20.Init() == true) && (mcp_0x21.Init() == true))
  {
    mcp_error = false;
  }
  else if ((mcp_0x20.Init() == false) || (mcp_0x21.Init() == false))
  {
    mcp_error = true;
  }
```

For some reason, the above code stops both MCP23017's from working all togeather (they no longer repond to library commads such as setPin and setAllPins). 

If I comment-out the above section of code, both MCP23017's function normally again.

Any advice would be greatly appreciated?

wollewald, 2023-12-24:
@RemyDavies the Init function checks the I2C or SPI connection, but it also resets your MCP23017. So, you should only call it once at the beginning. It checks the connection by writing a value into one of the registers and testing if it's really written into the register. Therefore it works with I2C and SPI. 
For checking the I2C connection I have just added another function called i2cConnectionError(). It returns the value from Wire.endTransmission. If zero, i.e. ``if(!mcp_name.i2cConectionError())`` then everything is fine. 
It's published as version 1.6.8. Please try.

RemyDavies, 2023-12-25:
The new fuction works perfectly - thank you for making such a quick update! :)

```
  if ((mcp_0x20.i2cConnectionError() == 0) && (mcp_0x21.i2cConnectionError() == 0))
  {
    mcp_error = false;
  }
  else if ((mcp_0x20.i2cConnectionError() != 0) || (mcp_0x21.i2cConnectionError() != 0))
  {
    mcp_error = true;
  }
```

The resetting of the MCP23017 when calling init() would explain the behaviour I was experiencing previously :)

wollewald, 2023-12-25:
Great! Thanks for the feedback.

closed 2023-12-25 by wollewald